andreika-git/hellen-one #298: megamodule-64

rusefillc, 2023-05-21:
получится ли
* can wakeup/sleep
* single or dual CAN bus?
* one uart for one LIN
в целом по входам - порядка 2chan плюс-минус?

Qwerty-OFF, 2023-05-21:
[image]
1x SWD (если шить через UART и забыть про TC2030, еще +2 IO)
2x CAN
1x USB
1x I2C
2x SPI (один можем обменять на 3 IO)
1x UART (есть возможность добавить один или два ценой по 2 IO)
1x WAKEUP
14x ADC/IO
12x IO (7 EINT)

В 2chan:
22x OUT
6x IN
12x ADC (из них 4 AUX)

Количество аппаратных ШИМ не отслеживал

Вывод: 40 полезных ног в 2chan, порядка 26-29 незадействованных ног в MM-64.

rusefillc, 2023-05-21:
1x SWD with TC2030
1x WAKEUP
2x CAN
1x USB

1x SENT input
2x UART
2x PWM?
14x ADC/IO
12x IO (7 EINT)
2x SPI
1x I2C (baro пока оставляем, но точно уже не монтируем. accel пока оставляем, но точно уже не монтируем)

Qwerty-OFF, 2023-05-21:
[image]
1x SWD
1x WAKEUP
2x CAN
1x USB
1x SENT input/DIN
2x UART
5x PWM OUT (один из них использует тот же таймер что и SENT)
4x OUT
14x ADC/DIN 
1x ADC VIGN (если надо, "бесплатно" превращается в полноценный ADC/DIN, так как наше количество каналов ОУ кратно четырем)
2x SPI
1x I2C

*Можно превратить 
1x SENT input/DIN
1x ADC
в
1x SENT input/ADC
1x PWM OUT

Qwerty-OFF, 2023-06-02:
1. Что делать с Board_id? Две ноги как раньше на это тратить не получается. Могу поставить I2C расширитель GPIO, как где-то когда-то пробовали раньше. Его лишние ноги можем использовать для решения проблем 2, 3, 4.
2. Если оставляем в модуле акселерометр, он потребует один пин CS. Избавляюсь от акселерометра?
3. SD карту оставляем? Тоже один пин CS.
4. На отладочные четыре светодиода не хватает ног.

rusefillc, 2023-06-03:
2. акселерометр точно убираем
3. SD карту оставляем
4. два отладочных светодиода - синий и красный
1. в таком варианте ^^^ хочется оставить просто двухногий Board_id всё-таки

Qwerty-OFF, 2023-06-05:
@rusefillc картинка для оценки получившегося количества периферии, жду отмашку делать так или менять что-нибудь
[image]

rusefillc, 2023-06-05:
@Qwerty-OFF красота!

rusefillc, 2023-06-05:
@Qwerty-OFF красота. Давай продолжать!

closed 2023-06-09 by rusefillc