andreika-git/hellen-one #301: Create new VR module using MAX9924

ElDominio, 2023-06-14:
@Qwerty-OFF 

Original design considerations for the current VR module were, in order of importance:
1)  MAX9924 was not assembled by JLCPCB, and usually not stocked
2) Problems with compatibility with low-count triggers

I propose there be a VR module that uses MAX9924 since now problem (1) is not a factor, and (2) can still be handled by using this module in conjunction with the already-existing module for low count patterns (by having both modules on one board)

In my experience, MAX992x chips are very good at conditioning/producing a useable output with almost all triggers without the fuss of manual thresholds, and without the pains that we have been experiencing with the current module (sometimes needs caps? sometimes doesn't?)

If it has the same module-size as the current VR solution, would it not be a plug-and-play affair to use on the current line of Hellen ECU's? If so, can it be done?

@andreika-git

rusefillc, 2023-06-26:
@Qwerty-OFF see interesting note at https://rusefi.com/forum/viewtopic.php?p=47921#p47921 mostly FYI

rusefillc, 2023-07-23:
@Qwerty-OFF we've realized that we forgot about something:

vr-descrete allowed us to have VR or Hall using two pinout pins and one MCU pin. The current max9924 is a loss of functionality since it's ONLY VR. At least for smaller universal boards we require VR/Hall 2 pinout pins solution like brought up on https://github.com/rusefi/alphax-8chan/issues/113

ElDominio, 2024-02-23:
This was done! And I love 9924

closed 2024-02-23 by ElDominio