andreika-git/hellen-one #318: unclear failure

rusefillc, 2023-08-02:
https://github.com/rusefi/stim/actions/runs/5734374341/job/15540458399

```
* Checking Python module 'NumPy': OK
* Checking Python module 'cairocffi': OK
* Checking module dependencies: ffi and cairo
* Checking ffi...
* Library ffi FOUND!
* Checking cairo...
* Library cairo FOUND!
* Checking Python module 'gzip': OK
* Checking Python module 'simpleparse': OK
* Checking Python module 'PyVRML97': OK
All checks done!
Processing digital_inputs board...
_XSERVTransmkdir: ERROR: euid != 0,directory /tmp/.X11-unix will not be created.
Removing old files of the board...
Creating digital_inputsa-bom_replace_digital_inputs-a.csv...
* appending the CPL with offset (0,0)...
Traceback (most recent call last):
  File "bin/process_board.py", line 264, in <module>
    append_cpl(frame_path + "/" + frame_name + "-CPL.csv", board_cpl, "0", "0", "0", 0)
  File "bin/process_board.py", line 113, in append_cpl
    with open(src_fname, 'rt') as src_f, open(dst_fname, 'a') as dst_f:
FileNotFoundError: [Errno 2] No such file or directory: '/digital_inputs/boards/digital_inputsa-bom_replace_digital_inputs-a.csv/frame/a-CPL.csv'
ABORTING!
```

what is this about? can we have better failure message to help users like myself?

andreika-git, 2023-08-02:
That's about empty board suffix, which is passed as a command line argument, so the whole command line is parsed incorrectly (arguments shifted)

rusefillc, 2023-08-02:
@andreika-git thank you! let me work around it in https://github.com/rusefi/stim but we need to improve usability