andreika-git/hellen-one #320: Failed to Merge gerbers maybe units?

rusefillc, 2023-08-02:
```
Reading data from inputs ...
Traceback (most recent call last):
  File "/usr/local/lib/python3.8/runpy.py", line 194, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/usr/local/lib/python3.8/runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "/hellen-one/bin/gerbmerge/gerbmerge/__main__.py", line 5, in <module>
    gerbmerge.main()
  File "/hellen-one/bin/gerbmerge/gerbmerge/gerbmerge.py", line 845, in main
    sys.exit(merge(opts, args)) ## run germberge
  File "/hellen-one/bin/gerbmerge/gerbmerge/gerbmerge.py", line [371](https://github.com/rusefi/stim/actions/runs/5739866688/job/15556441792#step:8:372), in merge
    config.parseConfigFile(args[0])
  File "/hellen-one/bin/gerbmerge/gerbmerge/config.py", line 387, in parseConfigFile
    J.parseExcellon(fname)
  File "/hellen-one/bin/gerbmerge/gerbmerge/jobs.py", line 716, in parseExcellon
    raise RuntimeError("File %s units do match config file" % fullname)
RuntimeError: File /digital_inputs/boards/digital_inputs-a/frame/inputs.DRL units do match config file

Where was an ERROR executing the script! Code=1
ABORTING!
Error: Process completed with exit code 3.
```

https://github.com/rusefi/stim/actions/runs/5739866688/job/15556441792

rusefillc, 2023-08-02:
@chuckwagoncomputing can we improve the export script? Assuming this is all in fact about units can we override units during export? if it's not trivial to override units during export can we just fail sooner with a very clear message asking for specific setting in the project?

chuckwagoncomputing, 2023-08-05:
Looks like this error was ignored in the gerber parsing, but not in the drill file parsing.
https://github.com/andreika-git/gerbmerge/blob/43eb62470565f5f408cf8094047624bc43f90183/gerbmerge/jobs.py#L388-L390

rusefillc, 2023-08-06:
@chuckwagoncomputing you mean "gerbermerg does automatic conversion" but drill processing does not do automation conversion not does it throw an error?

I'd say the nice way forward would be to attempt similar auto-conversion in drill processing?

rusefillc, 2023-08-08:
looks like this issue was about wrong/old files in the input folder

proper thing to do would be https://github.com/andreika-git/hellen-one/issues/321

closed 2023-08-08 by rusefillc