andreika-git/hellen-one #326: MM100

Qwerty-OFF, 2023-08-20:
Из модуля торчит наружу:
8x IGN / 5В-выход для управления VND10
8x INJ / IO
8x PWM_OUT / IN
14x аналоговый вход
2х цифровой вход через операционный усилитель (VSS, knock)
4х цифровой вход через триггер Шмидта
2x SPI / 8x IO / CAN2 + 2x IO + SPI
CAN
USB
I2C
2x UART / 4x IO
SWD
@rusefillc жду корректировок или одобрения

rusefillc, 2023-08-21:
@Qwerty-OFF а почему knock цифровой?
всё очень позитивно
SWD наружу на грани пользы, но вероятно мало влияет на всю картину, так что пусть торчит.

Qwerty-OFF, 2023-08-26:
Поправил, knock аналоговый, все остальное без изменений

Qwerty-OFF, 2023-08-26:
@Nadezhda-n сделал схему, можно трассировать. Футпринт модуля ММ100 нужно будет сделать по ходу работ. Цель запихнуть все это в минимальные возможные габариты

Qwerty-OFF, 2023-08-31:
@Nadezhda-n я добавил пару земляных отверстий в проекте чтоб починить DRC, подправил outjob.
Пофикси тут пад земли чтоб зазор по вертикали был 0.2мм
[image]
Все остальное выглядит хорошо

Nadezhda-n, 2023-09-01:
[image]

rusefillc, 2023-09-04:
@Qwerty-OFF надо заполнить https://github.com/rusefi/rusefi/commit/a1291cef18575ab7edc827957f991d7b20c49427 файлик теперь :)

Qwerty-OFF, 2023-10-02:
Сделано

closed 2023-10-02 by Qwerty-OFF