andreika-git/hellen-one #337: Error in gerber files with modules on the bottom side

Qwerty-OFF, 2023-10-08:
In 8chan-b and 8-chan-c, the WBO modules are placed on the bottom side of the board. There are artifacts on the top and bottom layers in the resulting gerber files. It seems that the artifacts are related to some issue with processing the rounded pads.
[image] [image]

andreika-git, 2023-10-10:
Fixed by https://github.com/rusefi/alphax-8chan/commit/2ebd618d90802731fe311708ef8f32424b1d1a04

andreika-git, 2023-10-10:
[image]

closed 2023-10-10 by Qwerty-OFF