andreika-git/hellen-one #341: mm100 f7

rusefillc, 2023-10-25:
F7 gives us so much needed RAM
mm100 gives us size

how does https://jlcpcb.com/partdetail/Stmicroelectronics-STM32F765VGT6/C435845 look for mm100-f7?

Qwerty-OFF, 2023-10-25:
Looks good
[image]

[image]

Qwerty-OFF, 2023-10-25:
Done

closed 2023-10-25 by Qwerty-OFF

rusefillc, 2023-11-22:
@Qwerty-OFF можно ли мне добавить в https://github.com/andreika-git/hellen-one/blob/master/modules/mega-mcu100-f7/readme.md файлик

* STM32F777VIT6
* STM32F767VIT6
* STM32F765VIT6

?

Qwerty-OFF, 2023-11-22:
Да, полный список пин-совместимых:
[image]