andreika-git/hellen-one #352: Wiki link broken power5 module

dynfer, 2023-12-14:
When accessing the link we get the following error

[image]

https://github.com/andreika-git/hellen-one/wiki/module-power5

rusefillc, 2023-12-15:
@dynfer please clarify where did you find that https://github.com/andreika-git/hellen-one/wiki/module-power5 link?

dynfer, 2023-12-15:
> @dynfer please clarify where did you find that https://github.com/andreika-git/hellen-one/wiki/module-power5 link?

Its the link in the modules section on the main page of the wiki, also I've noticed the Power12and5 module is missing in the list where it does exists in the repository(readme.md)

rusefillc, 2023-12-15:
thank you should be better now

closed 2023-12-15 by rusefillc