andreika-git/hellen-one #353: mm100 broken

rusefillc, 2023-12-15:


rusefillc, 2023-12-15:
снял два конденсатора - заработало. у меня 10 штук uaefi если нужна статистика, но что-то подозрение на проблему у нас

[image]

rusefillc, 2023-12-19:
[image]

Qwerty-OFF, 2023-12-19:
На линии 3.3В слишком большая емкостная нагрузка, стабилизатор уходит в защиту/перезапуск. Я повторил в мегамодулях модуль sd, в котором было 2 х 22 мкФ, если их убрать, все чинится. @andreika-git нет возражений если я в bom уменьшу эти конденсаторы и оставлю что-то около 10мкФ в сумме?

andreika-git, 2023-12-22:
https://resources.altium.com/p/how-to-design-microsd-power-circuits-without-destabilizing-on-board-voltage-supply

"The best value for a decoupling capacitor should exceed the total load capacitance and the capacitance between Vdd and Vss of the microSD card. In most cases, any values **above 45uF** will do."

andreika-git, 2023-12-22:
> @andreika-git нет возражений если я в bom уменьшу эти конденсаторы и оставлю что-то около 10мкФ в сумме?

Там большая ёмкость была не по моей прихоти, а согласно рекомендациям (см. коммент выше с примером рекомедации, но их таких много в других источниках). Идея, типа, в том, что с недостаточной ёмкостью, если вставить карту памяти "на горячую", то проц может перезагрузиться из-за скачка тока. Но, конечно, если выбирать между нестабильной работой в целом и броском при ините карты памяти, то придётся уменьшать ёмкость, да...

rusefillc, 2023-12-23:
@Qwerty-OFF need new PN for C117, C123

Qwerty-OFF, 2023-12-23:
> @Qwerty-OFF need new PN for C117, C123

Можешь убедиться что с одним снятым конденсатором все хорошо? Поставим 2x10 uF C77040. Либо один из тех двух сделаем DNP

Qwerty-OFF, 2023-12-23:
> @Qwerty-OFF need new PN for C117, C123

Можешь убедиться что с одним снятым конденсатором все хорошо? Поставим 2x10 uF C77040. Либо один из тех двух сделаем DNP

Нам критична установка флешки на горячую? Можно попробовать соорудить фильтр с катушкой. Хотя раньше оно же работало...

rusefillc, 2023-12-23:
> Можешь убедиться что с одним снятым конденсатором все хорошо?  Поставим 2x10 uF C77040. Либо один из тех двух сделаем DNP

да, с одним всё хорошо!

> Нам критична установка флешки на горячую?

вообще не критично

rusefillc, 2024-01-11:
placed an order with CL10A226MQ8NRNC C773 for C117,C123 and we are having the reboot issue

rusefillc, 2024-01-12:
> да, с одним всё хорошо!

похоже нет, не всегда всё хорошо с одним

Qwerty-OFF, 2025-05-03:
Кажется починил все актуальные модули и их проекты

closed 2025-05-03 by Qwerty-OFF