andreika-git/hellen-one #354: Output module - AUIPS2041LTR / C113610 no longer manufactured (out of stock)

dynfer, 2023-12-16:
The AUIPS2041LTR is no longer being manufactured. meaning its basically out of stock on all major resellers. A replacement would be needed.

Mfn PN AUIPS2041LTR
LCSC PN C113610 

This component is used in all 3 rev. of the module (0.1/0.2/0.3)

rusefillc, 2023-12-16:
> all 3 rev. of the module

@dynfer I can guess module name but overall let's spell it out?

See https://github.com/andreika-git/hellen-one/wiki/JLC-fabrication-component-substitution-options

See https://github.com/rusefi/uaefi/wiki/BOM-substitutions

TBD best way to document all those options

rusefillc, 2023-12-16:
> > all 3 rev. of the module
> 
> @dynfer I can guess module name but overall let's spell it out?

i am blind it's in the title

rusefillc, 2023-12-16:
I see some options mentioned at https://github.com/andreika-git/hellen-one/wiki/module-output#parts-options