andreika-git/hellen-one #376: TypeError: unsupported operand type(s) for &: 'float' and 'int'

rusefillc, 2024-03-26:
```
Job knock: 
  Extents: (0,0)-(36614,[1299](https://github.com/rusefi/hellen154hyundai/actions/runs/8376474943/job/22937135774#step:13:1300)2)
  Size: 0.366140" x 0.129920"

Trimming Excellon data to board outlines ...
Trimming Gerber data to board outlines ...
Traceback (most recent call last):
  File "/usr/local/lib/python3.8/runpy.py", line 194, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/usr/local/lib/python3.8/runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "/hellen-one/bin/gerbmerge/gerbmerge/__main__.py", line 5, in <module>
    gerbmerge.main()
  File "/hellen-one/bin/gerbmerge/gerbmerge/gerbmerge.py", line 845, in main
    sys.exit(merge(opts, args)) ## run germberge
  File "/hellen-one/bin/gerbmerge/gerbmerge/gerbmerge.py", line 408, in merge
    job.trimGerber()
  File "/hellen-one/bin/gerbmerge/gerbmerge/jobs.py", line 1184, in trimGerber
    self.trimGerberLayer(layername)
  File "/hellen-one/bin/gerbmerge/gerbmerge/jobs.py", line 1058, in trimGerberLayer
    newX, newY = geometry.rectCenter(newRect)
  File "/hellen-one/bin/gerbmerge/gerbmerge/geometry.py", line 280, in rectCenter
    if dx & 1:    # Odd width: center is (left+right)/2 + 1/2
TypeError: unsupported operand type(s) for &: 'float' and 'int'

Where was an ERROR executing the script! Code=1
ABORTING!
Error: Process completed with exit code 3.
```

rusefillc, 2024-03-26:
Некоторые куски гербера верхнего слоя меди подходят вплотную к границам модуля. Например, левый вертикальный пад 0.2х12.5 имеет нулевой зазор, и из-за конвертации из миллиметров в дюймы и погрешности арифметики он начинает "вылазить" за границы модуля на 0.000254 мм... Мы могли бы изменить формат работы с герберами на мм вместо дюймов, но у нас часть файлов в дюймах, а часть в мм, и потенциально могут вылезти проблемы в других модулях и фреймах... Или если бы Кикад умел экспортить герберы в дюймах, но он не умеет... Поэтому предлагаю обойтись малой кровью и либо немного увеличить границы модуля, либо чуть уменьшить/сдвинуть пады внутрь...

Qwerty-OFF, 2024-03-27:
Отодвинул на 0.01мм границы, теперь все работает