andreika-git/hellen-one #378: The SN74 is a bus transceiver/level shifter not a gate driver

rusefillc, 2024-04-05:
[image]

[Rafael Matos](https://www.facebook.com/groups/938729756561668/user/100088293504989/) says

50ma per output is definitely not plenty at all to drive an IGBT for ignition either inside or outside the coil doesn't really matter an ISL9v3040s might use around 70ma with 5v on the gate, I have no questions about it turning ON I know it will drive, but its not reliable, i never seen an ignition driver circuit like this straight from a level shifter, all the ECU's I have ever owned/repaired/designed use very similar ignition driver circuits with a few small differences but same concept, after the level shifter it's always good practice to have some gate driver that can handle more current.
The SN74 is a bus transceiver/level shifter not a gate driver

rusefillc, 2024-04-05:
@dron0gus @andreika-git any opinions? we use SN74 on all recent mega-module designs

rusefillc, 2024-04-08:
originally reported at https://www.facebook.com/groups/rusEfi/permalink/1876919989409302/

rusefillc, 2024-04-09:
[image]

This is scope trace of trigger input and ign output .
Does anyone know how big the pullup is on the ign output?
Edit: this was WITH coil connected. Volt is same scale, red reaches 5v, blue only 2,5v.

rusefillc, 2024-04-09:
huh

https://www.nxp.com/docs/en/data-sheet/MC33810.pdf

[image]

dron0gus, 2024-04-09:
Do we sure that all smart coils have IGBTs as power switch? May be some have bipolar transistors or even darlington?
Looking at oscilogram from https://github.com/andreika-git/hellen-one/issues/378#issuecomment-2045234004 I suspect bipolar or gate protection. Fronts are sharp I see no other reason to stop on 2.5V, does not look like output stage overload.

Another note. MC33810 is good for on-board IGBTs and dump (two-wire) coils. Definitely not the best choice for driving smart coils - no feedback benefits, no diagnostic.

dron0gus, 2024-04-09:
High voltage NPN for ignition coil https://www.st.com/en/power-transistors/std901t.html
[image]

rusefillc, 2024-09-15:
Some suggestions at https://rusefi.com/forum/viewtopic.php?p=49945#p49945

rusefillc, 2025-03-30:
``the current UAEFI uses a TI SN74LVC4245APWR Logic Translator for U103. It is capable of sourcing and sinking 24 mA on the outputs. Spent some time and found a pin-compatible TI SN74LXCH8T245PWR capable of source/sink at 50 mA.``

https://www.rusefi.com/forum/viewtopic.php?p=50475#p50475