andreika-git/hellen-one #385: kicad 8 does not seem to work

rusefillc, 2024-04-16:
https://github.com/rusefi/alphax-4chan/ with https://github.com/andreika-git/hellen-one/pull/384 has failed

```
Plotted to 'gerber/alphax_4ch-B_Silkscreen.gbo'.
Plotted to 'gerber/alphax_4ch-F_Mask.gts'.
Plotted to 'gerber/alphax_4ch-B_Mask.gbs'.
Plotted to 'gerber/alphax_4ch-Edge_Cuts.gm1'.
Export drill file
Loading board
Created file 'gerber/alphax_4ch.drl'
Done.
Export positions file
Loading board
Getting Drill/Place origin from PCB
aux_axis_origin is missing in the PCB file
Error: Process completed with exit code [25](https://github.com/rusefi/alphax-4chan/actions/runs/8709991859/job/23890877698#step:6:26)5.
```

rusefillc, 2024-04-16:
@chuckwagoncomputing can you have a look? maybe hellen scripts were not compatible with https://github.com/andreika-git/hellen-one/pull/384/commits/1cc79bec8c438d7a05ef36f1a818c5a984a02696?

rusefillc, 2024-04-16:
@chuckwagoncomputing reverted 8 for now

chuckwagoncomputing, 2024-04-17:
Oh I think this happened because the workflow calls the scripts in the submodule, and the submodule wasn't updated.
I tried it on my branch, and it failed in a different way, not sure if Kicad 8 related.
```
copy_from_Kicad.py: Error! Designator [R15] was not found in the BOM file!
```

https://github.com/chuckwagoncomputing/alphax-4chan/actions/runs/8713580962/job/23902272004

rusefillc, 2024-04-17:
let's see https://github.com/rusefi/alphax-4chan/actions/runs/8715382463

rusefillc, 2024-04-17:
@chuckwagoncomputing note that R15 appears in .cvs https://github.com/rusefi/alphax-4chan/commit/ca2d9e57c714975391fb0aecbeb029a404c7210a

looks like DNP is handled differently or smth looks like we would need @andreika-git

rusefillc, 2024-09-14:
Kicad 8 seems to be happy these days

closed 2024-09-14 by rusefillc