andreika-git/hellen-one #397: TVS Diodes quality issues

rusefillc, 2024-12-16:
* not all C85364 behave the same
* bad PN https://jlcpcb.com/partdetail/Msksemi-SRV05_4_PT7/C6454456

https://github.com/rusefi/hellen154hyundai/issues/163

dron0gus, 2024-12-17:
Я думаю, приподнимается вся линия V5A.
[image]

А нужно ли вообще цеплять пин REF (5) SRV05-4 на линию V5A? У него внутри встроенный стабилитрон на 6.0В, пусть он работает.
V5A у нас делает TLS115 который не может сопротивляться повышению напряжения на своем же выходе из-за внешней зпитки. А запитка через 4.7К резистор будет (14-6)/4700 = 1.7 мА. Мы либо должны с учетом этого нагружать линию V5A тупо резистором либо использовать другой стабилизатор, либо отцепить все защитные цепи от этой линии (что кажется самым логичным).
Кстати аналогичная запитка будет и через pull-up резисторы по линиям всяких CLT, но замыкание такого входа уже не штаная ситуцая.

dron0gus, 2024-12-17:
Преложение выше не актуально. Внимательно посмотрел схему hellen модулей - внутренние +5V и выходящие наружу +5VAS это разные источники. Надо бы посмотреть что происходит со внутренними V5A (+5VP) и попробовать нагрузить их.

Qwerty-OFF, 2024-12-17:
Оставил картинку где смотреть тут https://github.com/rusefi/hellen154hyundai/issues/163#issuecomment-2549582150
Даже если V5A поднимается, это не должно бы приводить к поднятию остальных IO-линий