andreika-git/hellen-one #398: RTC battery is drained

rusefillc, 2025-01-18:
@dron0gus says add a diode?

[image]

@dron0gu says remove a pointless diode?

[image]

rusefillc, 2025-01-18:
https://github.com/andreika-git/hellen-one/commit/c1a4955dffc6152309ed1d8d2e2cb981bbcdade4

dron0gus, 2025-01-19:
1N4148 has too high leakage current for this application.
With U410 LDO disconnected and no D107 STM32 consumes about 1.09 uA on Vbat input.
If 1N4148 is added on LDO output - current rises up to 1.9uA.
So reverse current through 1N4148 is 0.8 uA.
This is twice bigger than defined in DS. Fake part?

[image]

rusefillc, 2025-01-19:
progress see https://github.com/andreika-git/hellen-one/blob/master/modules/power_12and5V/readme.md

@dron0gus keeps suggesting something else

rusefillc, 2025-04-22:
fixed on 0.3 https://github.com/andreika-git/hellen-one/tree/master/modules/power_12and5V

closed 2025-04-22 by rusefillc