andreika-git/hellen-one #399: TVS drama

rusefillc, 2025-01-19:
```
8chan issues. Seems I have cross talk on a few analog inputs as well as some other weird behavior. For example, aux analog 1 (1A) mapped to MAP input reads raw MAP 0.00x volts with wire removed from socket. Measured voltage on wire with dmm, reads zero. Actual map sensor is not plugged into engine side of harness. So this all makes sense. If I insert the wire back into the connector at 1A, raw MAP reads some voltage, around 3 volts. Measuring terminal inside ECU case still shows 0.00x volts. And there is some cross talk on the TPS/PPS lines. Manually swing throttle body open, TPS reacts as normal, but one PPS channel tracks the TPS movement. Likewise, operate pedal and it tracks as normal but one TPS channel tracks the PPS.
```

rusefillc, 2025-01-19:
related to https://github.com/rusefi/hellen154hyundai/issues/163?

rusefillc, 2025-01-19:
@Qwerty-OFF can one just remove some of these SRV in case of emergency?

[image]

Qwerty-OFF, 2025-01-19:
Only if no voltage higher than 5V is applied to this srv05-4.